polyend/TrackerBetaTesting #578: 1.3.0 Can't get NTRQ to run in NES emulator

ilambert, 2020-12-24:
I know it's technically not a bug, and I know it's not an advertised feature, but the tracker nerd in me wanted this to work so badly. 

Would be interesting for someone with debugging access to figure what the problem might be ;)

mFoxRU, 2020-12-25:
First of all, I really appreciate that you added this feature. But it would be great if you improve this feature a little bit.

PULSAR doesn't work either and show this
[image]

mFoxRU, 2020-12-25:
@ilambert For now you can try cajoNES, it seems to work fine
[image]

Djeroek, 2020-12-26:
Hah running a tracker inside the tracker, how meta :)

Don't want to start a new topic, but seeing this is a thread on the nes implementation; currently I think the games list is limited to 100 entries, I guess the same like with the samples folders in the beginning. Only the games folder isn't automatically split into >1 folder when the file count exceeds 100 files afaics. Would be cool if it does for the games list as well! Also, what are the A & B options for in the same screen you can load games? Don't seem to do anything here. 

One more improvement imo would be a button combo to reset back to the game list view instead of having to reset using the power button.

jaap3, 2020-12-27:
@Djeroek please do open a new ticket, it's much easier for the developers to track one specific thing per "topic".

About the A&B options in the Games screen, I think they just serve as a hint/reminder for where the A/B buttons are when playing a game.

XANi, 2021-01-06:
Tested on FF3 (english translation from vimm.net), froze tracker in intro. And the intro sounded so nice when ran thru chorus on my mixer... which made me think, sampling the game I currently play would be *pretty* cool

tfspsound, 2021-02-17:
I know it’s a bit off topic but any suggestions for good safe ROM sites?

closed 2021-12-02 by Jacknametrouble