polyend/TrackerBetaTesting #582: Allow SD recording (and other samples) delete from Sample Loader

n4s, 2020-12-25:
Firmware 1.3.0: There is no way to delete a recorded sample (or any other file) in the SD card using the Sample Loader.  Please note that there is a way to delete samples from the Instruments list, but if you are browsing for instance the Recording folder the Delete key won't delete the files that are on the SD Card.  I think adding a button (replacing the Delete Unused if you are browsing on the left side) and/or implement the Delete key would be useful to discard bad recordings.

jaap3, 2020-12-27:
This has been requested before in #342

ketumatek, 2021-01-23:
What will be useful is the possibility to delete folder sample or sample in the SD card from the loader sample..
At this time you have to listen to your sample go back to your computer and delete sample...
I put a lot of folder sample on my se card that I don t really are in ...when I looking for a sample sometimes I have complete folder I don t want...so if I can immediately delete it it will be very cool and less wast of time...
Because actually I m listen my sample on the tracker notice the folder I don t want go to my computer delete them to have more space on my card (128G not a small one).
Think about it please...(I Know there are more urgent requests..but it will be better for work)

SeanHub, 2021-11-30:
Noticed this myself, I created a naff sample and would like to delete it without needing to remove the SD card and connect to a computer