polyend/TrackerBetaTesting #585: [1.3.0] File > Games with empty Games folder crashes Tracker

tilllt, 2020-12-25:
### Steps to reproduce
Tap Files, Tap Games -> Crash

tilllt, 2020-12-27:
https://youtu.be/UvU5q3BgYeU

phaelam, 2020-12-28:
perhaps if the games folder is going to be part of the install....it might be best there is something in it so it doesn't crash. I checked to see if anything came with the update. and in doing so, as its empty it crashed. no need for games...deleted folder

tilllt, 2020-12-28:
As mentioned here
https://github.com/polyend/TrackerIssues/issues/596

Apparently it doesn't crash, there is just nothing you can do in the games menu when you haven't installed the games. The bug is that you cannot exit the menu in any way.

phaelam, 2020-12-28:
Being stuck there not being able to do anything is equivalent to a crash.

ziej8asek, 2021-01-04:
Duplicate of #596

closed 2021-01-04 by ziej8asek