polyend/TrackerBetaTesting #586: [Firmware version] Stereo exactly the opposite in exported files

Konevitsaus, 2020-12-26:
This one has been reported already in the beginning of November but hasn't gotten any classification at all and has not been fixed. So, here we go again:

When you make stereo effects in the tracker, the exported file has them exactly in the opposite direction, always. For example: when a piano has been panned to the left speaker, the exported file has it in the right speaker.

I hope this will get fixed soon, since it seems like quite a big mistake. Thanks!

jaap3, 2020-12-27:
I think you are referring to issue #337. Interestingly enough the same user now reports that stereo information is gone in the latest firmware (issue #580).

What firmware version are you on currently?

doelman, 2020-12-27:
@jaap3 those are two seperate things. This topic is about export song, the other (#580) is about export stems.

Konevitsaus, 2020-12-30:
I am on the latest firmware version, but the issue has been there in all versions I've tried. And yes, I am talking about exporting songs into wave files.

shuler323, 2021-01-15:
thread moved to #697

closed 2021-01-15 by shuler323