polyend/TrackerBetaTesting #588: 1.3.0. Song Export does not work properly

DJadolfo, 2020-12-26:
## Hi, sorry you had trouble!

### Steps to reproduce

I  upgraded firmware to 1.3.0. beta 2 and found out that no more Song export works correct (patterns do not sound right, also slots does not follow in correct order). I went on with 1.3.0. official release and the things went worse - the song stucks on first slot without any sound.

Downgrading to 1.2.1 solved the Song export problem (except the 1.3.0 effects aren't working, of course). When upgrading to 1.3.0. beta 2 again - the Song export again works.

Conclusion: there is a SONG EXPORT bug in 1.3.0. beta 2 which I faced, and that went even more critical with 1.3.0. official relase.

### Attachments and links

Movie upladed to https://failiem.lv/u/8sk6yttuw (link valid two weeks),

### Other important information

* How often does this bug happen? (every time, often, "it only happened once") Yes, it happen with my first effort to Export Song in 1.3.0. beta 2. All other exports earlier were done with 1.2.1. succesfully. As I told downgrading to 1.2.1 and up to 1.3.0. beta 2 did help me to manage a succesfull export, however I am not going futher to 1.3.0. official now. By the way, also I experienced crash when opening the Games (after several restarts one game also started but I could not figure out how to exit Games mode). So I stay with beta 2 instead.

* Does this bug happen in previous firmware versions?

NO

* Does this bug happen in every project?

Yes, I tried also earlier projects, the same problem. The song in my  video is Legowelt Tracker Demo, which also has the same Song Export problem in 1.3.0.

DJadolfo, 2021-01-02:
Update on this problem:
I tried again an older project (1.2.0) to export as a Song.
When Tracker is upgraded to 1.3.0.b6 or 1.3.0. - the Export is not working (no sound during Export process, folder is created but empty).
When Tracker is downgraded to 1.3.0.b2 - the Export works but sometimes a cracking noise sound appears which also can be heard on exported wav file.
Tried another project, and another SD card. The same behaviour.

clickysteve, 2021-01-05:
Thank you for reporting this again. 

Unfortunately exporting stems and songs has major issues, and hasn't worked correctly for many release versions. A fix for the tempo/export audio being bust has been outstanding for a while:

https://github.com/polyend/TrackerIssues/issues/512

bgdnrz, 2021-01-12:
I am experiencing issues with exporting as well (v1.3.0). Some tracks export fine. Others simply show "Exporting" forever, even after the song time has elapsed. Thanks!

shuler323, 2021-01-15:
thread moved to #697

closed 2021-01-15 by shuler323