polyend/TrackerBetaTesting #591: {1.3.0} Song mode -> multiple issues when editing

anemlife, 2020-12-26:
These bugs are pretty hard to explain as they don't happen all the time.

Example 1: When I delete a pattern by mistake and then hit undo, the function act in a random way. Sometimes it works and bring back the pattern, but other times it doesn't work or it brings back another pattern and not the one I just deleted. I think something is wrong with the way PT handles things you delete in its memory (in song mode). That would be cool if the undo option in song mode could work flawlessly as i already lost a few patterns this way! (also maybe a redo function is missing there?)

Example 2: Sometimes I delete a pattern in song mode, then I go back in pattern mode and the deleted pattern is still present. (I double checked and I was in the right pattern page every time it happened) Then I switch to the previous pattern and I go back to the other one, and now the part I deleted is finally gone. This is a minor bug but I thought good to mention that some things are definitely behaving strangely in song monde)

Example 3: When trying to copy long sections (for example 8 tracks 4 patterns all at the same time) I also had weird results like few parts missing, etc... 

As these were all related to song mode and edit in particular I decided to make a single topic about it! These bugs all happened during the last days (and were already present in the betas as far as I remember). I will try to do a video next time it happens but these kinds of bugs are quite hazardous.

tilllt, 2020-12-27:
I can second this irregularities in song mode, here is a video that shows something strange is going on. 
- selecting an area
- deleting it
- cursor jumps to random line in pattern

https://youtu.be/bNvq_-f45Jw

anemlife, 2020-12-27:
> I can second this irregularities in song mode, here is a video that shows something strange is going on.
> 
> * selecting an area
> * deleting it
> * cursor jumps to random line in pattern
> 
> https://youtu.be/bNvq_-f45Jw

The issue you showed in your video are actually not bugs! You had the steps set to 4 so every time you copy/paste anything the cursor will go 4 steps further! You have to set the step to 1 in order to achieve "usual" copy and paste function with no jump 👍   try this

Also the bugs I was talking about in the topic are related to the song mode but not pattern mode :)

tilllt, 2020-12-27:
Ok, you are right it's not song mode related. As for the rest: IF this is intended behaviour, I don't understand the intention of this.

anemlife, 2020-12-27:
I think it is intended as it as been working like this for months! In my case I always switch back to 1 step after writing notes or whatever so it never really annoyed me but I understand it can feels disturbing sometimes

tilllt, 2020-12-27:
I think what bothers me (without trying to hijack your issue here) is that the "steps" do not make sense after a deletion process. I cannot believe this is intended behaviour. But to go back to YOUR issue, i am discovering so many bugs and inconsistencies - and bad UI design especially in the song editor, that I am seriously considering if this machine is really usable yet.

anemlife, 2020-12-27:
Yeah there are definitely many bugs that need to get fixed there. But, I've now achieved a few songs with the PT (especially with older firmwares) with almost no issues and a lot of fun! New firmware brings tons of possibilities especially for midi but it's not the most stable one I agree. I think the PT is already usable but need a few adjustments :)

ambv, 2021-01-04:
> i am discovering so many bugs and inconsistencies - and bad UI design especially in the song editor

@tilllt, if you find any concrete issues, please file them.

(BTW, as @anemlife is saying, the Step behavior is intentional.)

tilllt, 2021-01-04:
@ambv did you check the issues on github? i filed a LOT of tickets over the last weeks since i own my Tracker.

tilllt, 2021-01-04:
@ambv Well, if the intentional behaviour is not good UX, maybe it could be opened for discussion?

ambv, 2021-01-04:
Sure, we can discuss it but I doubt it will change in this particular case because the current behavior is more flexible. You can achieve what you want by setting Step to 1. If we made a change like you want, pasting a line every n-th step would be impossible.

tilllt, 2021-01-04:
@ambv i am specifically talking about the deletion process, i dont know if that came across. I dount that the (really useful) Steps behaviour has the same beneficial functionality when deleting stuff.

ambv, 2021-01-04:
Gotcha. Let's open an issue about Steps behavior during deletion.

tilllt, 2021-01-04:
> 
> 
> Gotcha. Let's open an issue about Steps behavior during deletion.

here we go https://github.com/polyend/TrackerIssues/issues/639

mroemore, 2021-01-14:
I can confirm all of the issues @anemlife listed as of v1.3.1b2. I just lost 30 mins of work due to trying to undo the deletion of a track in song mode, and it instead undoing completely unrelated work in another pattern. Then it crashed. Song mode crashes frequently, seems to happen more often if I copy/paste a few tracks then hit the Pattern button.

closed 2021-01-23 by ambv