polyend/TrackerBetaTesting #594: [1.3.0] (Beat Slice) Slices already written on pattern shift in consonance with the pad roots note.

LetsSeeIfWeSolveThis, 2020-12-26:
I was on C3 (pad root note) and Chromatic scale, I used beat slice mode and wrote the pattern as I wanted. 

But once I tried to select the scale of my song on project settings, from C3 **pad root note** I moved to C#3 (**+1 semitone**), then, for example, a Kick on C3 slice and Snare on C#3 slice moves to C#3 and D3 respectively. (**+1 slice**).

* How often does this bug happen? (every time, often, "it only happened once")
Everytime.
* Does this bug happen in previous firmware versions?
No.
* Does this bug happen in every project?
Yes.

LetsSeeIfWeSolveThis, 2020-12-26:
#589 Could be the same? He didn't mention beat slice mode but... sounds close enough.

PiotrAleksander, 2021-01-05:
@LetsSeeIfWeSolveThis It is not the same bug. 
Here, the beat slices are **moved up a semitone** which causes issue with already programmed patterns (e. g. kicks becomes snares and snares something else).
I've updated from 1.2 and had noticed it too. I think beat slices shouldn't be moved or all already programmed beat notes should be altered on root note change (does not sound feasible though).
@andrzejmagierski2 I preferred the old way, so after choosing scale I was changing beat slices with the jog wheel, but played notes with the pads.

LetsSeeIfWeSolveThis, 2021-01-05:
Yep, I rushed to the conclusion, thank you for pointing out!

closed 2021-01-07 by andrzejmagierski2