polyend/TrackerBetaTesting #596: [1.3.0] Can not exit Games menu

builderchad, 2020-12-27:
When in the Games menu (to select which game to play) I can not leave and go back to any other screen or menu (such as Song or Pattern) or even the preceding File screen.  The only way to continue is to power down the device (with the long "Shutting down..." progress bar).

When in the Games menu the only buttons that work are the up and down and "Start" buttons and the power button.  The "A" and "B" and "Select" buttons do nothing.  All of the other main buttons on the device are also inactive so I can not go back from the Games menu to the Pattern screen for example.

If I accidentally hit the Games key when in the File screen I must power down and restart the machine losing UNDO information.

### Steps to reproduce

1. Switch on device
2. Press "File" button
3. Select "Games" button (below screen)
4. Press any button to change mode/screen such as "Pattern"

### Other important information

Bug happens every time on 1.3.0 with every project

forestcaver, 2020-12-27:
I think it’s by design - you dont need to shutdown - just tap and release the “power” (reset) button.

builderchad, 2020-12-27:
Hi there.  Thanks for the reply.  My unit does not behave like that.  When I tap and release the the power button while in the Games menu it simply flashes up the "Shutting down" modal very quickly and then it disappears as soon as I release the button - and I'm still stuck in the Games menu.  Only a complete power down and power on gets me out.  Cheers.

tilllt, 2020-12-27:
I think this may be a duplicate of
https://github.com/polyend/TrackerIssues/issues/585

builderchad, 2020-12-27:
Possibly but it does not feel like a crash since the Games menu arrow keys and Start button still function normally and I can still launch games.

tilllt, 2020-12-27:
Then probably you are right and to me it just looks like a crash, since i didnt download the games and i cannot do anything in the empty games folder and not go back to any other mennu of the tracker. In any case its the same bug, i assume.

jaap3, 2020-12-27:
I can confirm this issue. It's not possible to exit the Games screen after entering it. In addition it's not possible to use the arrow keys to navigate the games list.

ssviatko, 2020-12-31:
Took delivery of my unit a few hours ago, and I can confirm this issue, the unit hangs when you enter the Games screen, requiring a poweroff/poweron.

MetabaronCZ, 2020-12-31:
Yes, I also confirm this. I think the problem is that someone forgot to add "Exit" button to the screen :). I also dont have any games, so I had to shut down the unit to get out of the menu.

But I think it should behave consistently, so clicking on Pattern or Song button, etc. should just work.

closed 2021-01-08 by ziej8asek

synthexpert, 2021-01-09:
How can I do to exit game mode? with the latest release I still have to power off. Is there a key combination?

SeanHub, 2021-12-03:
This isn't fixed on 1.4.0 either