polyend/TrackerBetaTesting #599: [1.3.0] Intermittent corruption in sample data in ram

tilllt, 2020-12-27:
### Steps to reproduce

Work on a Project, do something (havent figured out what precisely), Sample in RAM gets corrupted.
Restarting the tracker reloads the sample and fixes the problem

### Attachments and links

I dont think the error can be reliably reproduced by a certain action, but since it happened twice, its a reocurring issue.

bestenfallskongenial, 2021-01-12:
I've got it.

Guess that's why it happens:

I load a lot of samples with equal name over 16 characters long different only in the index number.
I applied some sample editor effects.
The sample I edited was merged into another Sample.

tilllt, 2021-02-23:
I dont think thats the problem i have...
But i could finally reproduce it, while actually trying to film another [Firmware Bug](https://github.com/polyend/TrackerIssues/issues/835), it occured again, i think it has something to do with connecting USB devices:

https://www.pixelscope.de/videos/watch/c73e5ea1-6708-46cf-9149-b70a9ce133c7

MetabaronCZ, 2021-02-23:
Hi. I see you are using volume FX. It is not relative to instrument volume. For my projects if I set it over value of 35, it beggins to  clip. Isnt that the problem?

tilllt, 2021-02-23:
This issue is a about sample corruption, so no, it has nothing to do with Instrument Volume FX.
This video shows it crystal clear: https://www.pixelscope.de/videos/watch/6a014b7a-0006-41af-ba32-4ab1fe7cf01d

MetabaronCZ, 2021-02-23:
Oh, I see. Thats really weird.

bestenfallskongenial, 2021-02-23:
Wonder how and why I got emails about this topic, but that's not the error I refer to.
My samples are merge into each other.
Sadly I don't do pictures or videos

tilllt, 2021-02-24:
> Wonder how and why I got emails about this topic, but that's not the error I refer to.
> My samples are merge into each other.
> Sadly I don't do pictures or videos

You get notifications because you posted this: https://github.com/polyend/TrackerIssues/issues/599#issuecomment-758832741

pete223, 2021-06-14:
Hey @tilllt, I've been able to reliably reproduce the issue you were having when I power my PT from a MacBook Pro 2015.
Would you be able to tell me what model of Macbook you have in the linked videos?

tilllt, 2021-06-14:
Ancient, Macook A1260, Late 2008. 

In a Discourse chat, @ambv mentioned that it could be something like a "brown out" where the tracker momentarily gets too little power while USB-Hub and Mac negotiate USB Power requirements, and thus the samples in memory get corrupted. I dont have the tools (or time) to check that theory but it sounds like a something that could very well be the cause. So, in theory, trying different usb hubs might be a workaround to the problem.

pete223, 2021-06-14:
As I posted in the linked issue, the problem also happens when I power the PT directly from my MacBook Pro. Maybe there is no fix for this if that momentary power drop is coming from the Mac

tilllt, 2021-06-14:
theoretically its possible that its not power related but that our Macbooks are sending something weird over USB which makes the tracker choke. But again, that would require a usb debugger to check, which i dont have. After all its pretty easy to work around that issue, if you know where its coming from (which took me a while to figure out).