polyend/TrackerBetaTesting #600: [1.3.0] Can't Shut down

thebradclay, 2020-12-27:
## Hi, sorry you had trouble!

### Steps to reproduce
I was previewing a sample. The sample played a strange noise then the samples disappeared from the sd card section, I could not back out via arrow keys. I attempted to power down but the Tracker would not power down and when the progress bar got to the end it would audibly click during the power down and then continued to power down over and over, see video.

Also, global changes to scale doesn't stay changed between new projects. It is always set at C3 and anti-alising on all new projects.

### Attachments and links

Video of tracker not powering down:
https://drive.google.com/file/d/11SK0yZZCiygOkoH4z382F3wicj_STWe2/view?usp=sharing

Project file
[dec262020.zip](https://github.com/polyend/TrackerIssues/files/5744750/dec262020.zip)

### Other important information

* it's only happened once  

* Does this bug happen in previous firmware versions? NO

* Does this bug happen in every project? Hasn't yet

ambv, 2021-01-05:
> global changes to scale doesn't stay changed between new projects. It is always set at C3 and anti-alising on all new projects.

There are no global scale and anti-aliasing settings anymore. They are per-project. Where did you set the global settings?

> Tracker would not power down and when the progress bar got to the end it would audibly click during the power down and then continued to power down over and over

Thanks for the video. We'll be looking out for it but it's not something we were able to reproduce so far.

ambv, 2021-01-24:
Closing due to no activity.

closed 2021-01-24 by ambv