polyend/TrackerBetaTesting #601: [1.3.0] File Transfer Mode Doesn't Work Anymore

imabigdipper, 2020-12-27:
### Steps to reproduce

* Connect Tracker to Windows computer, start file transfer mode, and try to access SD card from computer.
* The SD card will show up in file explorer, along with info on how full it is, but if you try to enter and explore the SD card it will say it is empty.

### Attachments and links
[image]
[image]
[image]

### Other important information

* Tested on multiple Windows 10 computers, happens on all of them

* How often does this bug happen? 
  * Every Time

* Does this bug happen in previous firmware versions?
  * No, file transfer worked fine on all 1.2.x versions for me. Haven't tried any 1.3 beta firmware though

Just-Peter, 2020-12-27:
The same on MacOS Big Sur on my Mac Mini M1. It worked on the betas with Android File Transfer, but not anymore on final 1.3.0

jaap3, 2020-12-27:
I believe this is similar or at least related to this issue #574

imabigdipper, 2020-12-27:
> I believe this is similar or at least related to this issue #574

Agreed, it seems to be universally broken. Although I wasn't sure if #574 was a MacOS specific issue. So I figured I would make a separate issue since I'm on Windows

tilllt, 2021-01-01:
Cannot confirm, works for me on Win10. I don't know if this step was not necessary before, so just to make sure: you did select config > file transfer > start file transfer and it does not show up?

imabigdipper, 2021-01-02:
> Cannot confirm, works for me on Win10. I don't know if this step was not necessary before, so just to make sure: you did select config > file transfer > start file transfer and it does not show up?

Yep, did config > file transfer > start file transfer. Once the tracker is in file transfer mode, the SD card does show up, along with its storage usage, as shown in my screenshot, but when you try to open it, it shows as empty. Maybe its an issue with my individual unit? I tested this on 3 different windows computers and all had the same issue.

tilllt, 2021-01-02:
... or maybe windows 10 version? I updated all of my windows laptops recently with the latest Win10 update..

draper-m, 2021-05-05:
Using win10. Just for completeness. Hope this helps.
-can enable the file transfer mode
-can view the sd card and its contents on PC although very slow to access folders that are then *empty
-can't transfer files - hangs on copying
-have experienced disabling ?file corruptions on sd card when moving files using dongle requiring reformatting of sd card
-*have lost preloaded artists projects and instruments despite transferring a backup folder of the orig sd card onto the sd via dongle
-win10 doesn't seem to approve polyend tool app.
...going to use PT as standalone for now until bugs fixed :-( Otherwise having great fun with it :-)

ROM72, 2021-06-15:
I'm in 1.3.0, from time to time when I copy a sample from my pc (windows 10) straight to the tracker in transfer mode I can't find it at the end of the copy.

closed 2023-04-26 by MiroPoly