polyend/TrackerBetaTesting #603: [1.03]render selection line in audio garbled

shaddart, 2020-12-27:
## Hi, sorry you had trouble!

audio is garbled when using render selection to render audio from external synth that is being midi sequenced by tracker using the midi din out.   

### Steps to reproduce

select some external midi tracks in pattern mode , connect external synth to line in, do render selection. 

### Attachments and links

### Other important information

* happens every time

* Does this bug happen in previous firmware versions? Don’t know
* 
### 

https://user-images.githubusercontent.com/76660618/103177469-c3843a00-4848-11eb-9145-b28aa39ca5a7.MOV
[image]

https://user-images.githubusercontent.com/76660618/103177501-13fb9780-4849-11eb-8dbd-3b1b2ceca7b2.MOV

* Does this bug happen in every project? Both that I tried

db9979, 2020-12-27:
Yeah this is also my problem. 
Firmware 1.3.0
Happens every time. Sound is brilliant by direkt monitor through - only after rendering it crackles totally.

shaddart, 2020-12-27:
Did it work before with any of the betas?

jazzycreates, 2020-12-28:
I am also having this issue! 
Firmware 1.3.0
I just got the tracker a few days ago so have not had any experience with other firmwares or betas.
I tried different projects, different external synths, different cables, turning it off and on again, placing the "cursor" on the four paramter types (note, instrument, fx1, fx2) just in case, including an audio sample within the same track (the audio sample comes in clear as day, but the MIDI notes do not after rendering the selection), headphone outs vs line outs. No dice. Garble audio every time.

shaddart, 2020-12-28:
I found out you can use the sample recorder page while playing Midi, just hit play once you get in there and it will record your external midi instruments through the inputs  just fine, you’ll just need to crop it

jazzycreates, 2020-12-29:
Sweet! Thanks so much for letting me know!

jazzycreates, 2020-12-31:
I had put in a ticket to Polyend and with the response I got back it seems that this might not be a bug, but a feature that does not have support (yet?). While I had thought this was a MIDI out only issue, it seems that it's actually because we are trying to render audio from the external line in port while the sequencer is playing and I thought I had read somewhere that that wasn't a feature they planned on implementing. I didn't understand what it meant until now.

shaddart, 2020-12-31:
weird that it works in the sample recorder - you can play midi and record the line in at the same time - so should be able to in pattern render

eimhin, 2021-01-02:
I opened feature request #483 for this a while back. It was a beta and “render selection” just rendered silence for midi tracks (with line-in audio). Recording external instruments from the sample recorder page seems to be the only way for now.

closed 2021-01-07 by shuler323