polyend/TrackerBetaTesting #604: Games

tfspsound, 2020-12-27:
When I first got my Tracker I never once thought I hope they make it play very basic games when I want a break from making music. When I found out they had, I wasn’t too bothered either. Not to detract from the hard work from put in from the people involved it’s great but I would have liked to have seen something like one of the following. 

A canvas for drawing a pixel based wave table or single cycle synth that you could export to use on a track for example. Something like mario paint but you draw a waveform then hit play. 

LAN has musical potential if instead or rerouting a network you could re route sounds pulses in a network to create melodies. I imagine notes falling in columns like in guitar heroes (and the Tracker) but we can deflect them, maybe use the jog wheel to alter the angle of the deflectors. The deflectors play when struck by the notes and there’d be different deflector types. We’d control the release of the notes and the pitch and where and at what angle our deflectors are at.  The idea is the deflections make interesting polyrhythms or semi random arpeggios. 

How about a drum synth for making drums sounds using random radio snippets that we could filter and combine and cross modulate again to make sounds for projects. 
Anyway didn’t mean to sound ungrateful, it’s good work I just think you’re maybe missing an opportunity. Would like to hear others thoughts if they have any?

jaap3, 2020-12-28:
Note that the currently the Games section is a NES (Nintendo/Famicom) emulator, so the possibilities are somewhat limited to whatever is possible on a NES. I believe the games that can be downloaded from Polyend are homebrew roms, and have not been developed specifically for the Tracker.

There's a fair amount of (homebrew) NES roms available on the internet. You might find something that suits your tastes (watch out for some of the more shady rom sites though ;-)

tfspsound, 2020-12-28:
Thanks Jaap I had no idea it was a NES emulator. Makes it slightly cooler but I still won’t be downloading anymore games unless there is maybe an NES version of LSDJ but even then I don’t think I’d bother. I can’t imagine how long it would take someone to programme my suggestions but I imagine years now if it has to be NES emulator compatible. Oh well 
Can we just delete the Games folder once it’s on the card or will that cause problems?
Cheers

jaap3, 2020-12-28:
I think you can just leave the Games folder empty, I don't think it's required to have any files in it. Just don't use the "Games" button on the file menu, as it currently cannot be exited (#596).

closed 2020-12-29 by tfspsound