polyend/TrackerBetaTesting #605: [1.3.0] display issue

phaelam, 2020-12-28:
after going into page 2 of Instruments Parameters, and returning to page 1
[image]

Jacknametrouble, 2020-12-28:
Also, transitions between Instrument Effects and Instrument Parameters is "moving" the amount pole from Automation to the Instrument Effects screen.

tilllt, 2020-12-28:
@Jacknametrouble could you outline the process at polyend on how bugs are prioritized and processed? I am a bit surprised that two bugs in a plethora of bugs received the bugs confirmed tag, others, which are older, didn't.

ziej8asek, 2021-01-04:
Duplicate of #595

ambv, 2021-01-04:
We'll be fixing it as part of #595.

closed 2021-01-04 by ambv