polyend/TrackerBetaTesting #610: Suggestion: Games -> saving and loading

anemlife, 2020-12-28:
The Game function is really great! But could you add any way for saving the current game state?

This would be really cool when travelling, I want to play a few games on the PT as it feels pretty comfortable but would really love if I could also end them without having to play 5 hours straight!

Thanks :)

keith-macleod, 2021-01-01:
It would be really cool if the emulation supported roms that save their state to cartridge. Same reason, I guess, but right now, it seems as though these games won't even run.

theorize999, 2021-02-17:
Hmm maybe that's why I can't get Final Fantasy to run... That sure would be cool to have save states, there are so many great NES games, PLUS it would open up room for a crafty programmer to do some really neat tracker related programs. Honestly I'm looking into what it takes to make homebrew ROMs... I would love to program an app or two for the tracker, something useful like a calculator for samples or theory tool, like a chord and scale dictionary. Heck it would be really awesome to make tutorials that people could just load up, forget about youtube, we could have a NES set of tuts complete with interesting characters! Also it would be really satisfying to make a fun game too! I'm not a programmer but I know some extreme basics, but I have a feeling I'll be learning some programming for NES soon!

ClaidArremer, 2021-04-09:
Being able to save and load would be amazing, both functions could have a dedicated button.