polyend/TrackerBetaTesting #611: [1.3.0] Upgraded from 1.3.0b2; reverb no longer sends if instrument volume is 0

captaincfcs, 2020-12-28:
## Hi, sorry you had trouble!

### Steps to reproduce

I have a project I was working on in beta version 1.3.0b2 that featured a loop of notes, the step effects of which made the volume 0 and the reverb 76 (see exhibit a). This gave the loop a soft and omnidirectional sound drenched in reverb. The master output shows that the reverb was sending, despite all step volumes being 0 (see exhibit b). However, when I updated to the official 1.3.0 release, the reverb no longer sends (see exhibit c). I can change the volume of the notes from 0 in order to hear them, but this drastically changes the sound that the loop had in version 1.3.0b2 (play attached example project in both 1.3.0b2 and 1.3.0 and you'll hear what I'm talking about).

### Attachments and links

**Folder of exhibit images a, b, c**
[exhibit images.zip](https://github.com/polyend/TrackerIssues/files/5748284/exhibit.images.zip)

**Project that showcases issue**
[project example.zip](https://github.com/polyend/TrackerIssues/files/5748285/project.example.zip)

### Other important information

* This bug happens every time I open this project in official release of 1.3.0

* This bug did not happen in previous firmware, to my knowledge.

* This bug seems to happen across projects, where applicable.

ambv, 2021-01-04:
There's a setting in Config called "Sends mode". If that setting is set to post-fader, reverb will only send if the instrument volume is higher than 0. If you want to achieve the previous behavior, set it to pre-fader. Let me know if that solved your issue.

captaincfcs, 2021-01-04:
It works! Thank you so so much.

closed 2021-01-04 by captaincfcs