polyend/TrackerBetaTesting #613: [1.3.0] Some freezes in Song Mode and Config

LetsSeeIfWeSolveThis, 2020-12-28:
Don't know how to reproduce it but **I was copying and pasting tracks between slots (the white bricks on song mode)** and I had a crash, concretely twice doing this. A constant tone keep sounding while crashing. Could not reboot via power button. Remove power wire and solved.

It also happened in config window.

sebpmtl, 2020-12-29:
> Don't know how to reproduce it but **I was copying and pasting tracks between slots (the white bricks on song mode)** and I had a crash, concretely twice doing this. A constant tone keep sounding [...]

I had the same thing happen to me 3 times since yesterday.

youngestbrother, 2020-12-29:
This is a recurring issue for me, has happened countless times

anemlife, 2020-12-31:
Same here, this happens really often. Song mode causes multiples crashes (it was the same in the betas as far as I remember)

tilllt, 2020-12-31:
https://youtu.be/5RmSctskxOs

tilllt, 2020-12-31:
As a side note: is there a QA/QC process at polyend? I am frankly a little surprised by the amount and severity of bugs in a release version of a firmware.

thebradclay, 2021-01-01:
Same issues. Multiple hard crashes yesterday. Lost several patterns. Beyond frustrated at this point.

benc-uk, 2021-01-02:
Same, had several crashes in 1.3.0
Often in song mode or config view

thebradclay, 2021-01-02:
Same issue here. 2 times yesterday.

bichan17, 2021-01-03:
i have also crashed on the song page. needed to unplug and replug.

thebradclay, 2021-01-03:
Just happened again, 3rd time today.

ambv, 2021-01-04:
This is our top priority to fix alongside #630. If you ever find a combination of steps that reliably always freezes the device on a given project, please share as it will speed up solving the problem a lot.

JimSomniac, 2021-01-04:
Am using an external hardware sequencer (Yamaha QY100) for clock and transport. Can hard crash the Tracker 100% of the time in any project.
1. enter song mode
2. press play on sequencer -song will play and sync fine
3. press "return to start" on sequencer - Tracker will completely crash. Have to pull the plug.

No such issue in pattern mode. "return to start" works fine.

I can post a video tomorrow if useful...

Thanks
Jim

closed 2021-01-05 by wjakobczyk

wjakobczyk-polyend, 2021-01-05:
I was able to reproduce and fix the problem as described by @JimSomniac above. 
Specifically this addresses crashes when doing play/stop from external midi/usb while in Song mode.
Some other instability issues are also discussed here, if they are not related to this scenario, please report separately.
Thanks!