polyend/TrackerBetaTesting #616: Support for microtonal tunings

Ahornberg, 2020-12-29:
I'm familiar with the software of Goattracker, Famitracker and LSDJ. All three have a frequency table built in to set the frequency for each note. For the named trackers, there exist some scripts to change the frequency table according to the tuning defined in a given a Scala .scl file. This would be the easiest way to make the firmware 'microtonal'.

Of course, it would be a more sophisticated solution to store the microtonal tuning for a given song in the song itself and not in the firmware/software of the tracker, but this would require some GUI to enter the data for the specific tuning.

For me, it would be helpful to know, where the frequency table is placed in the firmware for writing a script to feed my microtonal needs.

bgdnrz, 2021-01-05:
Pardon, not at all experienced in this field. But are you finding the combination of tune/finetune and fiddling around in wavetable/granular to be too limiting? Curious to hear what you're aiming for, just from a bystanders' pov. If not too much trouble to explain. :)

Ahornberg, 2021-01-05:
> Pardon, not at all experienced in this field. But are you finding the combination of tune/finetune and fiddling around in wavetable/granular to be too limiting? Curious to hear what you're aiming for, just from a bystanders' pov. If not too much trouble to explain. :)

Nearly all sequencing software is based on dividing an octave into 12 equal parts, the so called Standard Western Tuning, also known as semitones and measured in cents (1 semitone = 100 cents, 1 octave = 1200 cents).

Microtonalists can be divided in 2 groups:

One group stays on 12 notes per octave but uses e.g. Just Intonation (a pure fifth has 701.96 cents in contrast to 700 cents in Standard Western Tuning, a pure major third has 386.31 cents in contrast to 400 cents, a putre minor third has 315.64 cents). Usually this can be achieved by using the Microtune FX.

The other group of microtonalists, me included, uses a different number of pitches per octave or doesn't use octaves at all (e.g. the Bohlen-Pierce Tuning divides a duodecime = 1901.96 cents into 13 equal steps). My preferred tunings are dividing 2 octaves into 33 equal steps (so-called 33ed4), and dividing one octave into 94 equal steps (so-called 94edo). Common used tunings are 22edo, 31edo, 41edo, 53edo and 205edo (used by a midi controller called the Tonal Plexus). For us microtonalists, fiddling around with tuning parameters on a per note basis is simply a nightmare.

The easiest workaround would be to change the frequency table in the firmware by using a script that can read Scala files http://www.huygens-fokker.org/scala/ and change the frequency table in the firmware accordingly. For me, one way is to detect the frequency table by analyzing the firmware and by trial and error (a time-consuming process) or to ask here for detailed informations about the location and format of the frequency table inside the firmware.

For further informations see https://sevish.com/2016/microtonal-music-trackers/

bgdnrz, 2021-01-06:
Thank you for elaborating so extensively. Digging further. :)

jamesneville, 2021-03-15:
Came here to +1 for micro-tunings/alternative scales

loktez, 2021-04-01:
Clearly it could give to the tracker an other dimension, and if the midi could send these microtuning scales in a MPE format : oh my god, you kill the game.

craighitchings, 2021-09-11:
Any word on this coming to Tracker any time soon? I would love to use Just Intonation and other microtuning scales with Tracker

closed 2021-12-02 by Jacknametrouble

jdlesemajeste, 2021-12-27:
Hello, I would also like to put my vote in for improving microtonal support. Adding the ability to load and use scala files to map the button pad would be great. Thanks!