polyend/TrackerBetaTesting #617: [1.3.0] Dial knob lagging input on Euclidean Events bar

MegaTurrican, 2020-12-30:
When you enter the "Fill" menu, select "Euclidean" and move the selection to the "Events" tab. Now start adjusting values fast with the dial knob. You will notice some lagging and not smooth transition of values.

Happens every time.

wjakobczyk-polyend, 2021-01-04:
I can observe a log of less than half a secod, when turning the wheel really fast. Is it the same that you observe?
Then I guess it's not a big issue...?

MegaTurrican, 2021-01-04:
Indeed, it's not that big of issue; It's just that in that specific tab the lagging is more noticeable than others (try to "play" with fast dial changes in other tabs too). I thought I'd mention it, just in case it is something unexpected.

wjakobczyk-polyend, 2021-01-04:
Thank you, I don't think it's an indication of something more, so we'll leave it as is.

closed 2021-01-04 by wjakobczyk-polyend