polyend/TrackerBetaTesting #621: [1.3.0] CUT function clips (like cuting an audio file without a tiny fade out)

LetsSeeIfWeSolveThis, 2020-12-31:
I'm noticing this using the CUT function.

doelman, 2020-12-31:
Yeah I'm experiencing this problem a lot as well. Not sure if it's clipping but a lot of ticks and crackles. Very annoying

closed 2021-01-05 by ambv

doelman, 2021-01-05:
@ambv I get what you're saying but this clipping / crackling /ticking does not happen consistently. Sometimes it doesn't happen at all and when you play the same pattern again it does. So there must be something buggy as well that causes the sound to clip /crackle sometimes. Also, when using the roll-effects this happens way too often (but also not consistently). I'll attach a project example in a sec.

doelman, 2021-01-05:
[gf-mix1.zip](https://github.com/polyend/TrackerIssues/files/5770553/gf-mix1.zip)

@ambv 
check the guitar intro (track 8 of the first pattern in the song) for example. Listen to it on solo, a few times and you'll hear these crackles are there but not consistently. If you have another suggestion on how to fix this within the current possibilities, please let me know but I tried several things and it keeps doing this. Also with other samples.

jasonfnorth, 2022-05-27:
If this is by design, what is the recommendation to stip playback of a sample before starting another and not introduce clipping?  Use FAD?