polyend/TrackerBetaTesting #623: [1.3.0] Midi issues/Freezing/losing patterns

thebradclay, 2021-01-01:
I have midi out to Mutable Instruments Yarns, from Yarns Midi Out to Roland TB-3. I've also had the same happen when directly outputting midi to the TB-3. I have tried with multiple cables.

I made a short video attached below. I changed patterns and entered a new note. I then tried to ad midi cc. nothing happens. 4th time this has happened today. Also this is in a new project with nothing else going on. I couldn't power down via the power button and had to unplug the unit.

In another project the same thing happened but when I switched patterns all the notes/fx etc were no longer there in 3 out of 6 patterns. 

https://user-images.githubusercontent.com/73397631/103432092-ee9fbe00-4ba7-11eb-8d49-9ab7e749617b.mov

[New Project.zip](https://github.com/polyend/TrackerIssues/files/5758549/New.Project.zip)

* How often does this bug happen? 4th time today

* Does this bug happen in previous firmware versions? Not that I recall

* Does this bug happen in every project? So far it's happened in two projects today.

wjakobczyk-polyend, 2021-01-04:
What actually happens next? It's not clear in the video - has the device hanged, or just not responding to the jog when editing CC value?

thebradclay, 2021-01-04:
@wjakobczyk-polyend in the video I am trying to choose an fx. it shows that you can scroll to one but not choose it.

wjakobczyk-polyend, 2021-01-04:
Does the a/b/c/d/e/f show up in pattern view? (the video is too low quality to see it).
Does this happen for any MIDI channel/instrument?
Could you show the Instrument Parameters screen for the instrument?
Does this also happen for a new project setup in this firmware version?

Sorry for a long list of questions, but we are not able to reproduce the problem so need to nail what is causing this specific to your setup. Thanks!

thebradclay, 2021-01-04:
Does the a/b/c/d/e/f show up in pattern view? (the video is too low quality to see it).
Yes  "a/b/c/d/e/f " were visible in the fx list but when scrolled to and then release the button it would not load nor could you scroll to it.

Does this happen for any MIDI channel/instrument?
Yes

Could you show the Instrument Parameters screen for the instrument?
MIDI channel 2 is what I had it set to.
The only MIDI parameters I had set were CCA 74 (this is for cutoff of the TB-3) and CCB 71 (resonance).

Does this also happen for a new project setup in this firmware version?
It happened in a couple more. I have reverted back to the working beta.

I have attached the project file above.

wjakobczyk-polyend, 2021-01-04:
Thanks a lot!
One, hopefully last question - this is not happening in a beta release? Which one are you using?

thebradclay, 2021-01-04:
@wjakobczyk-polyend 
Hey no worries! 
I apparently miss spoke/typed. I am still on the current 1.3.0 firmware but could have sworn I downgraded. Currently midi is working in this project. The one I attached is the only one I have that was giving issues.

thebradclay, 2021-01-04:
I did just go and save the current project I am working on and the save name I have been using was not present when I hit save, it allowed me to "save as" without any name being input. I went back and resaved as the appropriate name and it asked if I should overwrite that project name, which is the one that I was attempting to save when it didn't populate the name field.

wjakobczyk-polyend, 2021-01-05:
These issues might be caused by switching between releases (especially going back from 1.3.0 to an earlier beta) and something getting messed up in configuration and/or projects. We have not designed (nor tested) migration procedures for moving backwards.

I suggest you stay on 1.3.0 and hopefully this works ok (though there may be issues with specific projects if they got somehow corrupted because of the switching).

FWIW I'm still not able to reproduce the MIDI CC problem even with the project you attached, using latest release.

thebradclay, 2021-01-05:
@wjakobczyk-polyend This project I am speaking about was made entirely under the current firmware.

MortalJS, 2022-03-19:
No response. Closed

trutypesounds, 2022-06-02:
@MortalJS disappearing patterns are random and is still an issue in 1.6 betas, had a couple of patterns went empty after exiting performance mode