polyend/TrackerBetaTesting #624: 1.3.0 [Firmware version] MOD file import

ssviatko, 2021-01-01:
MOD files don't import properly/this aspect of the software needs some work.

To reproduce this bug: Download an old Amiga MOD file made 30+ years ago. A great example mod file can be found here:

https://archive.org/download/Heatbeat-TrackerModules/StreetJungle.mod

Copy this file into the "Modules" directory on the SD card, press the "File" button, and "Import Mod". Allow the PT to import the file into a new project.

Play the MOD. The volume levels of the various instruments are all wrong. It does not sound at all the same way it does on the Amiga. Even a modern version of VLC will play the mod properly.

Video of the MOD playing properly: https://youtu.be/RGz6-QeM_2s

benc-uk, 2021-01-03:
The manual does state this feature is experimental

https://polyend.com/manuals/tracker-manual/#loading-saving-projects---importing-exporting-projects
_"Note: The .mod file import is still under the development and its functionality will be improved."_

k4ot, 2021-06-18:
i second this
just got the tracker yesterday
so i thought "lets do music from the future? HELL NOES ! lets play amiga mods"
what a surprise to load the enigma demo mod and hear the playback is total verruckt and not any single instrument plays right

moshier1, 2021-12-04:
Agreed. This was part of the reason I bought one, I assumed importing and exporting mod files would work well.

closed 2022-02-06 by MortalJS