polyend/TrackerBetaTesting #626: Project details

bgdnrz, 2021-01-01:
It would be very useful if it was possible to see project details in the File > Open dialog. For example, the currently selected project would show (in a pane similar to Fx details for example):

- project length (mm:ss)
- project creation date
- project last edit date
- total samples

benc-uk, 2021-01-03:
I'd like to see project length too, but I think calculating the length/time night not be easy, as the tempo fx can modulate the tempo dynamically so it wouldn't be possible to know the length without rendering the whole track :(

piotr3500, 2021-01-04:
Unfortunately, we do not have a reference clock inside, so no way to tell the dates.
We might add a text field, that would be saved to txt file for each project and previewed while choosing projects.
Would it help?

ketumatek, 2021-01-04:
> Unfortunately, we do not have a reference clock inside, so no way to tell the dates.
> 
> We might add a text field, that would be saved to txt file for each project and previewed while choosing projects.
> 
> Would it help?
> 
>  

I think I was ask in another tread...I should be a nice idea

bgdnrz, 2021-01-05:
> Unfortunately, we do not have a reference clock inside, so no way to tell the dates.

Interesting! I read about some techniques utilizing the CPU clock cycle which made me wonder if you could somehow generate a pseudo timestamp of sorts. Something to the effect of "total time passed since the project was opened" which then gets saved to the project and incremented on subsequent opens. But it's starting to sound like a fiddly thousand piece Jenga board, so scrap the idea!

> We might add a text field, that would be saved to txt file for each project and previewed while choosing projects.
> Would it help?

Yes! I am finding myself versioning tracks, so a "notes" field would be a great help.

Kintaro, 2021-01-06:
No reference clock means there is no absolute clock/date internally. Every time you power the unit on, the clock basically starts from 0. So there would be no way to know how long it has been since the project was opened.

ketumatek, 2021-01-06:
> Unfortunately, we do not have a reference clock inside, so no way to tell the dates.
> 
I think what it means by dates is not the current date but the dates you make the track for exemple 06.01.2021
So notes will be enough to write this..

x2db, 2021-04-15:
> No reference clock means there is no absolute clock/date internally. Every time you power the unit on, the clock basically starts from 0. So there would be no way to know how long it has been since the project was opened.

well this is actually not a problem. If you have a counter that increments any time you save a project and then store this number per project, you could sort by this number, so essentially the oldest project would be on the bottom and the most fresh one (highest counter number) would be on the top, I wish Piotr added this to the 1.4 too:)

x2db, 2021-04-15:
> Unfortunately, we do not have a reference clock inside, so no way to tell the dates.
> We might add a text field, that would be saved to txt file for each project and previewed while choosing projects.
> Would it help?

Piotr, see my note, counter that increments when we save could solve sorting by date, then only issue I see when you change the project stamp/date on a computer, but then again anything saved on tracker would use this counter number to sort by last edited