polyend/TrackerBetaTesting #628: Suggestion: When renaming files -> option to paste the last name used

anemlife, 2021-01-02:
When renaming instruments or whatever, it would be nice if we could "recall" the last name used. 

Example: 

- I rename an instrument as "hi-hat"
- I go to the next instrument to rename it as well
- There is an option down the keyboard called "recall last name" I press it
- The name "hi-hat" is recalled
- Now I can simply add a number or letter to keep organized like "hi-hat b"  

An alternative could be an option to copy/paste what is written on the keyboard!

This would allow for a faster workflow!

NoraCodes, 2021-01-25:
I would also really appreciate this - a similar interface would be really useful when sampling. For example when recording some variations on a bass drum patch I would like to enter "modular kick" once and then just enter details after pasting, like "modular kick resonant hi", "modular kick punchy low", etc.

closed 2021-04-27 by anemlife

NoraCodes, 2021-04-27:
May I ask why you closed this @anemlife?

anemlife, 2021-04-27:
I'm closing every requests I made that they will never add/care about (in my opinion) 

There are hundred of requests and the firmware updates seems more and more rare lately so I try to keep only my "crucial" requests as I don't think their team will be able to carry all of them. 

There are things that are completely killing the workflow of the PT at the moment IMO and I would like them to notice these in priority. 
I can still reopen these "secondaries" requests one day but for now I need other functions to work correctly! 

Hope you understand

@NoraCodes

NoraCodes, 2021-04-27:
I think a group of professional software developers are probably plenty capable of prioritizing issues, but I do understand what you're saying. I'd argue it should remain open and you should note it as not as important to you.

anemlife, 2021-04-27:
Oh yeah they're probably capable of prioritizing, no doubt about that.

But by looking at the number of requests I think they will have to take decisions about things they add or not so I prefer if they work on more important options 
(for example: preview working when PT is slaved, export + rendering working when PT is slaved, access to every octaves without going to config, easier navigation in pattern modes, C0 to C10 playable on pads, sine lfo, lfo offset, no glitchy noises when exporting stems, transpose FX in sample editor, performance FXs to affect MIDI instruments, etc etc...). 

All of these feel way more crucial to me! That's why i will keep smaller requests closed for the moment