polyend/TrackerBetaTesting #629: 1.3.0 - Disable option in limiter sidechain doesn't disable the limiter

Intothegroove, 2021-01-02:
When choosing from the sidechain list of tracks in the limiter, choosing disable at the top doesn't disable the limiter. If you reduce the limiter threshold and change the attack and release, it still works and effects the whole project it sounds like. I think it's a great Idea though to keep this option to effect the whole project as well as each individual track but maybe just rename it from "Disable" to "All" sidechain threshold?

* How often does this bug happen? every time

* Does this bug happen in previous firmware versions? No

* Does this bug happen in every project? Yes

doelman, 2021-01-02:
@Intothegroove maybe I'm wrong but I would assume "disable" disables sidechaining and not the limiter itself. I'm not sure what you mean with sidechaining from all? 

If you don't want the limiter to be effective just leave the threshold at highest level or set the attack time to a very high level.

Intothegroove, 2021-01-02:
> @Intothegroove maybe I'm wrong but I would assume "disable" disables sidechaining and not the limiter itself. I'm not sure what you mean with sidechaining from all?
> 
> If you don't want the limiter to be ineffective just leave the threshold at highest level or set the attack time to a very high level.

You're right! lol. Sorry about that, I was thinking of something else. Thanks for clearing that up :-)

doelman, 2021-01-04:
@Intothegroove Good to hear it's clear now. You can close the issue then, I guess

closed 2021-01-04 by ambv