polyend/TrackerBetaTesting #630: [1.3.0] fatal errors -> makes the PT useless for live

anemlife, 2021-01-03:
Had so much bugs in the last few days when trying to record songs live (song mode switching to performance mode making the PT freeze almost every 2 minutes) that I came to the conclusion that 1.3.0 makes the PT barelly not usable for live (and I'm a bit confused as it is an official release, there are soooo much bugs in this one). 

Don't get me wrong I love the tracker but the fact that I can't even record a song without being scared of making the unit freeze (and maybe lose datas at the same time) makes the PT pretty painful to work with for concrete results! 

I know there are several topics about that but I just wanted to point the fact that performance mode and song mode are crucial to make the PT work 100% and the fact that these are still so buggy is pretty annoying so I definitely hope that next firmware releases will focus on solving these issues before adding tons of features (even if I like it!) 

Hope this topic doesn't feels wrong :) have nice holidays and thanks for the great work 

### Steps to reproduce

PT slaved to ableton by USB MIDI
Play a few patterns in song mode 
Switch between performance and song mode while it's playing
PT often freezes

### Other important information

* How often does this bug happen? (every time, often, "it only happened once")
* Every time

* Does this bug happen in previous firmware versions?
* in 1.3 betas yes, not so often with previous releases 

* Does this bug happen in every project?
* yes

tilllt, 2021-01-03:
Yeah, for me it's not only the bugs which are annoying but also UI things which are not completely thought through, counter-intuitive or just not implemented. And I posted a lot of very simple "enhancement" issues in that regard here.

It feels a bit as if somebody really wanted to release a new firmware (and the successive press release mentioning the new features, to boost sales) before the end of the year and cut beta testing and quality assurance very short to reach that schedule. Firmware 1.3.0 is not release quality software. I don't know about the versions before since I just got my PT, but this one is very disappointing.

And I know it was busy end of the year time with xmas and new years eve and covid on top, but there should be some beta testers that could at least confirm some of the bugs and make them official so the users feel they will be worked on (soon).

modulationsource, 2021-01-03:
Yes, the feature List has grown very fast, thank you for that. 
But I also think it is time to remove the bugs, we can wait for new features...

anemlife, 2021-01-03:
Yeah same feeling here, maybe 1.3.0 was released a bit too fast (compared to previous firmware where we had tons of betas before an official release). As far as I remember the official 1.2.0 release had way less bugs than this one!

I love all the new functions but would prefer if the next firmware could focus on making the basics functions usable without all these crashes! Sidechain, midi CC, etc... are great adds but I feel like the priority should now be pointed on making all the existing functions + UI stuff work 100%

Just-Peter, 2021-01-03:
I think they rushed the 1.3.0. release....(If I remember well) because they promised to release it before the end of the year. Just a few beta releases were not enough, there still were too many bugs.

tilllt, 2021-01-03:
Well in fact a software doesn't magically turn release quality just because you stop calling it a beta :) 1.3.0 is more like "alpha" quality of you apply the logic of other software projects.

anemlife, 2021-01-03:
I simply was surprised that official 1.3.0 firmware feels so much like a beta version compared to when they released the official 1.2.0 that felt pretty solid after the multiples betas (1.16/17/etc...)

As you guys said they probably had to rush it for the end of the year, and it was still a good Christmas present! 
But I thought it was important to point out the fact that basic functions still don't work flawlessly while we get tons of new features. For people looking to use the PT for live performance it's essential to have a device that won't freeze every 2 minutes and I feel more and more worried about this as none of the firmware releases seems 100% reliable about freezes at the moment.

Intothegroove, 2021-01-03:
I totally agree too. We need the basic functionally to work properly before introducing anything new, with every new release something that was working properly and is a main function gets messed up. Please Polyend, just focus on getting the firmware solid and usable before adding more :-(

thebradclay, 2021-01-03:
I have to agree with everyone else here. I love the tracker. My absolute favorite piece of gear. Having said that, this new firmware has caused me so many headaches that I don't enjoy using it as much. In the last few days I have had so many hard freezes that I've lost count. The horrible rendering makes finishing a project a nightmare. The midi issues, song mode freezing, patterns randomly disappearing and so many more. I really don't understand why this was released.

ambv, 2021-01-04:
Hi there, fam. We're back after the holiday break. Sorry to hear you're having problems. This issue, along with #613 is our top priority to solve at this point. We'll be focusing on bug fixes for the next few weeks so we'll get this under control.

To help us with solving this as soon as possible, if you ever find a problem that reproduces **every time**, please send it over to us with a list of things you're doing that trigger the issue, e.g. 1. load this project, 2. go to this screen, 3. press this button, 4. boom - there's a problem. Assigning zipped project files is **very** helpful. Short videos demonstrating the problem are also **very** helpful.

Thanks for sticking with us, please stay tuned for the first beta of 1.3.1 on Friday.

anemlife, 2021-01-04:
Hi! Thanks for getting back, hope you had great holidays!

Just did a video to show a way to make the PT crash almost every time: https://www.youtube.com/watch?v=QkbJ-1tH_ng&feature=youtu.be&ab_channel=Titi

Here's the way to reproduce:
1) open any project
2) MIDI config --> Clock in/USB + Transport in/USB + CC in/USB
3) go to the song mode screen
4) start playing on Ableton (or any DAW I guess) --> song starts with no issue
5) go to Performance mode
6) select every channel for performance FX (they turn red)
7) change patterns for all tracks via MIDI CC (using Nanopad to Ableton then to Tracker in my video)
8) go back to song mode 
9) pause the song on your daw
10) boom! PT freezes like 80% of the time (at least in my case and this goes for any project!) you can hear a strange glitchy sound at the end of the video

I'm not linking the project as this happened with basically any project since I'm using 1.3.0. but I can if needed. Thanks!

closed 2021-01-05 by wjakobczyk

wjakobczyk-polyend, 2021-01-05:
I was able to reproduce and fix the problem as described by @anemlife above. 
Specifically this addresses crashes when doing play/stop from external midi/usb while in Song mode.
Some other instability issues are also discussed here, if they are not related to this scenario, please report separately.
Thanks!

clickysteve, 2021-01-05:
> This issue, along with #613 is our top priority to solve at this point. We'll be focusing on bug fixes for the next few weeks so we'll get this under control.

@wjakobczyk @ambv 

I am pleased to see that Polyend are active here and responding to issues, but https://github.com/polyend/TrackerIssues/issues/512 has been outstanding for some time, and it doesn't seem to be listed as a priority.

Can you advise on the roadmap for when that will be tackled? Export is completely broken, and that is a major blocker to the machine being usable. The rendering of stems for export has never worked properly, as far as I'm aware, and there's been a few different bug reports which have been 'fixed' but still not worked - so the issue is far older than the ones reported here.

It is a bit disappointing to see new features being added, new bugs being created (and then fixed), but still no solution to what is a major issue with the core functionality of the Tracker.

thebradclay, 2021-01-05:
@clickysteve I started another bug report about exporting earlier #644 . Hopefully this will get resolved because I can't keep using the Tracker if export isn't properly fixed.

clickysteve, 2021-01-05:
> @clickysteve I started another bug report about exporting earlier #644 . Hopefully this will get resolved because I can't keep using the Tracker if export isn't properly fixed.

Yup. I've not been able to use the Tracker at all since I bought it, because exporting stems has never worked properly. There's been audio issues, and then tempo issues. It's currently an expensive paper weight, and I'm getting frustrated. I can't even return it at this point because it's past the return period. :/ I wish I had realised there was such significant problems with the export, as I wouldn't have bought it.

ambv, 2021-01-05:
@clickysteve we will be fixing performance issues with export in upcoming releases. We hear you loud and clear.

thebradclay, 2021-01-05:
@ambv Just for clarification, when you say "upcoming releases"....is this sooner rather than later?  Because right now exporting any of my current projects is just not working for me. Even recording in real time is out of the question because of drift.

ambv, 2021-01-06:
@thebradclay what is the extent of realtime playback drift you're experiencing?

thebradclay, 2021-01-06:
@ambv it varies but is audibly noticeable. Next time I sit down to record something I will try to figure out a way to capture video.

ambv, 2021-01-06:
@thebradclay "audibly noticeable" suggests to me you're comparing it against another sound source. Any reason why you're not using a common clock? Setting Tracker to follow Ableton's external clock sounds like it would help your problem.

thebradclay, 2021-01-06:
@ambv I did not say I was using another sound source. I said it was audibly noticeable. Meaning I hear the difference. What makes you think there is no common clock? Why would I set the Tracker to follow an external clock when I am recording the direct output of the Tracker?

clickysteve, 2021-01-06:
> @clickysteve we will be fixing performance issues with export in upcoming releases. We hear you loud and clear.

Thank you. I appreciate the quick reply. FWIW, I am glad that you are using GH and so transparent/active in the threads here. It makes a big difference.

thebradclay, 2021-01-06:
Agreed with @clickysteve being able to see the issues handled helps.