polyend/TrackerBetaTesting #631: Micro SD card stock contents

Kintaro, 2021-01-03:
*Before you start, did you check whether your feature was already requested?*
Yes I did and couldn't find anything.

*Please tell us about your feature in as much detail as you can provide.*
In case the original SD card contents are lost and let's say the backup also fails or someone buys a second hand unit, it would be good to have the original stock SD card files available for download. Either via the website or as part of the firmware repository.

jtkthx, 2021-01-03:
This document: [https://github.com/polyend/TrackerIssues/blob/master/HOWTO_BETA_FIRMWARE.md](https://github.com/polyend/TrackerIssues/blob/master/HOWTO_BETA_FIRMWARE.md)

Contains this link:  [https://polyend.com/files/Tracker_SD_1_3.zip](https://polyend.com/files/Tracker_SD_1_3.zip)

Kintaro, 2021-01-03:
Oooh, ok, that's a bit hidden. Probably good to put that on the website.

closed 2021-01-03 by Kintaro