polyend/TrackerBetaTesting #633: 1.3.0 - Sample recorder volume lower again

Intothegroove, 2021-01-03:
With the release of 1.3.0, it seems that the preview volume of a recorded sample is lower than previous firmware versions again. No matter how loud the signal is recorded, when played back it's lower than before. 

* How often does this bug happen? (every time

* Does this bug happen in previous firmware versions? No

* Does this bug happen in every project? Yes

thebradclay, 2021-01-03:
I noticed this last night while recording from the radio. It was louder when I previewed it on the record page. Went to the editor page and had to amplify it.

tilllt, 2021-01-03:
The radio recording volume has always been super low on my device, I usually have to crank up gain completely to get a decent level. Before I bought the tracker I watched some reviews of it and it already had the problem there is i fear it's just a bad implementation of the SDR circuit.

ambv, 2021-01-05:
@Intothegroove, let's clarify some things:
* Sample Recorder's "gain" setting controls how much your Line In/Mic/Radio will be amplified in the sample;
* "Preview" in the Sample Recorder is always the same volume as the recorded sample, perceived volume is only controlled by Master 1/3 Volume;
* "Preview" in Sample Loader is always the same volume as the recorded sample, perceived volume is only controlled by Master 1/3 Volume;
* "Preview" in Sample Playback, as well as when the sample is played by the sequencer or by you using pads, is affected by the Instrument's volume (Instrument Parameters 1/2), track mixers volumes (Master 2/3), and by Dry Mix (Master 3/3), and finally by Master 1/3 Volume.

So, if you record a sample and then it's quieter than you expect, check all the listed places for lowered volume.

Intothegroove, 2021-01-06:
> @Intothegroove, let's clarify some things:
>
>     * Sample Recorder's "gain" setting controls how much your Line In/Mic/Radio will be amplified in the sample;
>
>     * "Preview" in the Sample Recorder [...]

I have checked everything you said and there is definitely a lowered volume when previewing the recorded sample. I even went back to firmware 1.2.2 and absolutely there is a difference.

ambv, 2021-01-06:
Do you mean the "Preview" button during sample recording? Are you recording the radio or a different source?

I have two units, tested with several radio stations and mixer settings and cannot reproduce your problem.

thebradclay, 2021-01-06:
@Intothegroove Yes agreed.

@ambv the sound is lower.

Intothegroove, 2021-01-06:
> Do you mean the "Preview" button during sample recording? Are you recording the radio or a different source?
> 
> I have two units, tested with several radio stations and mixer settings and cannot reproduce your problem.

Yes, so when you record a sample and then preview the recorded sample right after, it is lower in volume than the actual live audio source being monitored before recording. Regardless of it being sampled from the radio, line or mic input and if the gain both internally or externally is turned all the way up, it was fixed in the previous firmware's becasue for awhile the recorded sample volume was extremely low but was fixed after a few firmware releases. I'll try and get a video recorded and posted as soon as I get a chance. Thanks so much.

closed 2022-02-06 by MortalJS