polyend/TrackerBetaTesting #636: 1.3.0 - Instrument stops playing when switching pages

Intothegroove, 2021-01-03:
When playing an instrument either with trackers pads or external midi controller and switching pages (example: holding down notes and switching from the pattern page to the instrument parameter page to adjust the filter) The Instrument is cut off and has to be re-triggered to play the sound in the new page. It should keep playing no matter what page you go in and out of on the tracker. 

* How often does this bug happen? every time

* Does this bug happen in previous firmware versions? Yes

* Does this bug happen in every project? Yes

ambv, 2021-01-04:
This is a current limitation that we might remove in a future version as an enhancement.

Intothegroove, 2021-01-04:
> This is a current limitation that we might remove in a future version as an enhancement.

That would be fantastic! Thank you guys so much!

closed 2021-12-02 by Jacknametrouble