polyend/TrackerBetaTesting #639: RFC - Rethink Pattern Editor "Steps" functionality in terms of UX

tilllt, 2021-01-04:
I find that there are some design decisions regarding the "Steps" functionality in Pattern Editor which are counter-intuitive even if working as intended. 

- maybe it makes sense if "Steps" should be ignored for delete actions
- i think it should be discussed how steps behaves when reaching the pattern limit, should it automatically wrap to the start of the same pattern? should it spill over to the next pattern, what if there is no next pattern, auto create one? should it just stop?
- both could be optional behaviour that could be enabled / disabled in config

To give an example what i find to be very confusing behaviour, which even seems like a bug, in how it is implemented right now, doing a delete action with steps 4 enabled, after the delete the cursor endes up at a seemingly random line.
https://youtu.be/bNvq_-f45Jw

tfspsound, 2021-01-05:
I set up a thread a while ago about making Step, Step. By this I meant that if you are using Step the cursor will move the number of steps it’s set to, even if you are using the cursor keys. 
Totally agree with you on this, and also find the current system counter intuitive. If step is set to 4 and you hit delete you should be moved 4 steps on, same with the cursor keys. When I set step and the cursor buttons didn’t work I was surprised when Del did. 
I would also prefer Step to be a toggle / latch as opposed to having to set it. 
Not sure about about spilling into other patterns I like the way you edit just the pattern you are on, as it currently works. My only concern would be that as I work on a project my patterns don’t always remain in sequence, so if I had a spill option and left it on I might affect a pattern I didn’t mean to. By no means opposed to it as an option though.

XANi, 2021-01-27:
> I set up a thread a while ago about making Step, Step. By this I meant that if you are using Step the cursor will move the number of steps it’s set to, even if you are using the cursor keys.

You kinda have to have option to move with single steps for cursors, else how would you pick place to start ?

tfspsound, 2021-01-28:
That’s why is suggest the Step screen button becomes a latching switch. When it’s off it is as if Step is set to 0, when it’s on the cursor moves according to the step setting. It also mean that the values of Step could corresponded to their ‘correct’ position (8 steps would be button 8 instead of 9.)

closed 2021-12-02 by Jacknametrouble