polyend/TrackerBetaTesting #640: Additional/Exclusive Midi Tracks

eatcodeplay, 2021-01-05:
I would love to see additional Midi Tracks, if performance allows it. 

This would allow you to use the 8 existing tracks fully for samples/internal instruments but then you could still expand with external devices.

This would essentially put our beloved tracker on par with the Digitakt, which also allows 8 internal tracks and 8 midi exclusive tracks.

XANi, 2021-01-06:
MIDI Chords already made my Christmas (altho I'd prefer musical names instead of numbers) but this would turn the tracker into basically perfect device for me. Splitting off MIDI and non-MIDI effects would also help ergonomics a bit, and maybe MIDI page could even have MIDI specific effects like say MIDI LFO or MIDI Echo

danapeks, 2021-01-06:
Adding my name to this as it's my number one request above all else now. Would be a gamechanger in how useful this device is, and doesn't _feel_ like it should add too much more to the processor if just sending Midi messages (though i am not a developer!). I would guess this functionality would significantly increase sales too, as would just make the tracker so much better at contolling external studio gear whilst also sequencing samplers and allow it to fully compete with Digitakt etc.

benc-uk, 2021-01-06:
This would be an absolute game changer

I'm a developer and done work with MIDI - it's a very old protocol (40 years!) so takes up very little resources (back when processors ran in Mhz and not Ghz). However in terms of UX and internal changes it would still require a lot of work, but I'd love to see this

tilllt, 2021-01-09:
> However in terms of UX and internal changes it would still require a lot of work, but I'd love to see this

Not necessarily if the 8 midi channels get their own page as suggested above.

switchal, 2021-07-31:
This would be great and indeed next level above the digitakt!

eholzbach, 2021-08-22:
I'm still hoping for this feature. I became frustrated with 8 tracks shared between samples and midi chords I went back to a Roland MC-50 and a sampler. I spent a ton of time with FastTracker and Impulse Tracker twenty+ years ago. Neither were hardware, but both supported more than 8 tracks on pc's of the time. An additional set of midi only tracks would be ideal.

XANi, 2021-11-22:
yeah FT2 had 32 audio tracks and complex envelopes to boot

MetabaronCZ, 2021-11-22:
Well, Akai MPC One has 128 MIDI tracks + 8 Audio tracks and is only a little more expensive.

If people from Polyend can optimalize code, I would rather see simple internal synth engine. Even if it has one oscillator and filter. Or rework current filter to be not that shitty :).

closed 2021-12-02 by Jacknametrouble

XANi, 2021-12-03:
Is that "close, won't implement" @Jacknametrouble ?

sd-ditoy, 2021-12-03:
> Is that "close, won't implement" @Jacknametrouble ?

I'm equally as curious about this.

tilllt, 2021-12-03:
I find this "closing issues without any comment" behaviour totally weird. I mean how hard is copy pasting "implemented", "won't implement" or even a "not possible, hw constraints", just to give people a heads up in what's going on.

danapeks, 2021-12-03:
They've already said on the Facebook group and the Discord they don't want to do this. Real shame.

XANi, 2021-12-03:
Why have a public bug tracker in the first place...