polyend/TrackerBetaTesting #641: Save collections of instruments as 'kits' to load in to projects

seanlclarke, 2021-01-05:
I realise this can be done by just making blank projects and saving them with instruments in, but it would be great if we could save 'kits' of say XX samples that could be loaded in to a project in the next XX slots. People can then make 'drum kits' for example that can be loaded in with one click without overwriting other samples...

modulationsource, 2021-01-06:
This would be nice. Would also open the possibility to preserve instrument information.
Imagine sampling a synth on different notes, apply semitone offst, loops and filter options and save them as kit.
This would be a kind of mutlisample.

LetsSeeIfWeSolveThis, 2021-01-08:
Absolutely

seanlclarke, 2021-01-08:
I pretty much always start by loading 4-5 drum samples that work well together  (kick, snare, hat, clap, crash etc) on every project, so being able to load say 'D&B Kit' would be a real time saver  and hopefully not too hard to add ('save instruments as kit' option and a load 'kit' option in the samples page.

sprottenkopf, 2021-02-01:
Good idea! I was myself thinking in line of saving instruments as separate instrument files. The way its done in Milkytracker and Fasttracker. Those files include all settings and audio files.

Jacknametrouble, 2021-12-18:
This has been added in the newest 1.5b1 beta.

closed 2021-12-18 by Jacknametrouble