polyend/TrackerBetaTesting #643: [1.3.0] Root note question

phaelam, 2021-01-05:
I have a MIDI note set C5, is show up in my MIDI monitor as C3.

the root note is set in the tracker to C3, so that makes sense. but its there any way to get C3 in the trackers window to match what's being sent?

loktez, 2021-01-07:
Same for me ( Ableton Live...) 
So the max could be G12?
 my best

Jacknametrouble, 2021-01-08:
This is probably the matter of the used MIDI convention.

piotr3500, 2021-01-08:
It's not technically a bug but rather a convention.
https://www.midi.org/forum/830-midi-octave-and-note-numbering-standard

Either way, we decided to make it configurable.

sastani, 2021-01-15:
I believe this is the same as my issue...every note is two octaves off. I would consider this a bug, because of the following:

If I connect an external synth to Tracker’s MIDI in and out, then play C3 on an external synth in “live mode”, during playback it will actually trigger C5. Likewise, if I played F#5 it would playback F#7.

ambv, 2021-04-27:
Fixed in 1.4.0b1.

closed 2021-04-27 by ambv