polyend/TrackerBetaTesting #644: [1.3.0] Stems not exporting correctly

thebradclay, 2021-01-05:
### Steps to reproduce

Create a song and export it.

### Attachments and links
Here is a "song" I created just to show this export issue. The song contains 2 patterns, 3 samples and tempo changes. The exported stems do not maintain any of the Tempo changes. I also exported the song which does not contain the tempo changes.

I have included the project files, the stems and the rendered song .wav file:

[PT Crash Tests.zip](https://github.com/polyend/TrackerIssues/files/5772347/PT.Crash.Tests.zip)

### Other important information

* How often does this bug happen? (every time, often, "it only happened once")
every time
* Does this bug happen in previous firmware versions?
Yes there are export issues in previous firmware versions

* Does this bug happen in every project?
YES

clickysteve, 2021-01-05:
Thank you for reporting this again. Just noting that this has been an ongoing issue in different forms across various firmware revisions: 

https://github.com/polyend/TrackerIssues/issues/512

I hope that Polyend will prioritise this fix, as it's a core feature which has been completely broken for some time.

thebradclay, 2021-01-05:
@clickysteve literally every single project I have exported has had issues. I'll keep reporting them till it's resolved. I really love the machine and enjoy using it but this problem is making me have second thoughts.

closed 2021-01-06 by Jacknametrouble

thebradclay, 2021-01-06:
@Jacknametrouble Why is this closed?

Jacknametrouble, 2021-01-06:
@thebradclay this is a duplicate

clickysteve, 2021-01-06:
> @Jacknametrouble Why is this closed? @clickysteve you seeing this?

Don't fret - it's common to close duplicate reports to manage the response to the issue in a single place. Usually whatever report was earliest. :)

thebradclay, 2021-01-06:
@clickysteve , right on. Just frustrated as I have a rather large project that I really need the stems from for a streaming event that is coming up.

clickysteve, 2021-01-07:
> @clickysteve , right on. Just frustrated as I have a rather large project that I really need the stems from for a streaming event that is coming up.

As a workaround, if you go back to the version before 1.3.0 (before the full rebuild), there were still tempo issues, but they were in time with themselves (at least for me), so you could warp them slightly in Ableton and they would work. They were slightly off from the stated tempo, but might be worth a look.