polyend/TrackerBetaTesting #646: [1.3.0] Sustained notes do not loop and are cutoff

benc-uk, 2021-01-06:
It is impossible to hold a note indefinitely and loop it forever with the Tracker.  
This is a severe musical limitation as you are unable to put in a long pad sound, or drone into a track. And makes the Tracker difficult to use with ambient, drone, and experimental music

### Steps to reproduce

Sustain a note using a looped sample, after a number of seconds the sustain is released, even without a OFF or other note.
The number of seconds it takes is dependant on tempo, at 200BPM it will release after 18 seconds, lower tempos lasts longer but it can never hold forever

This behaviour happens with MIDI notes also

### Attachments and links
A simple project to reproduce the issue it attached
[sustain-bug.zip](https://github.com/polyend/TrackerIssues/files/5777453/sustain-bug.zip)

Other people have verified the issue here
https://www.facebook.com/groups/polyendtracker/permalink/409707253697586

### Other important information

* How often does this bug happen? - Every time

* Does this bug happen in previous firmware versions? - Unknown

* Does this bug happen in every project? - Yes

Just-Peter, 2021-01-06:
I agree with @benc-uk Forever looping is crucial for ambient/drone music. Even on a tracker this is a essential feature.

doomglue, 2021-01-07:
+1, switching off envelopes does not help neither as that seemed to be suggested as a fix a few times.

shuler323, 2021-01-08:
Thanks for reporting. Next beta (1.3.1b1) should fix this problem, tested on highest tempo for ~10mins. If not please reopen this issue

closed 2021-01-08 by shuler323

Just-Peter, 2021-03-31:
> It is impossible to hold a note indefinitely and loop it forever with the Tracker.
> This is a severe musical limitation as you are unable to put in a long pad sound, or drone into a track. And makes the Tracker difficult to use with ambient, drone, and experimental music
> 
> ### Steps to reproduce
> Sustain a note using a looped sample, after a number of seconds the sustain is released, even without a OFF or other note.
> The number of seconds it takes is dependant on tempo, at 200BPM it will release after 18 seconds, lower tempos lasts longer (approx 1min at 120BPM) but it can never hold forever.
> 
> This behaviour happens with MIDI notes also.
> 
> ### Attachments and links
> A simple project to reproduce the issue it attached
> [sustain-bug.zip](https://github.com/polyend/TrackerIssues/files/5777453/sustain-bug.zip)
> 
> Other people have verified the issue here
> https://www.facebook.com/groups/polyendtracker/permalink/409707253697586
> 
> ### Other important information
> * How often does this bug happen? - Every time
> * Does this bug happen in previous firmware versions? - Unknown
> * Does this bug happen in every project? - Yes

Is this implemented yet?

shuler323, 2021-04-01:
in v1.3.1b1 or above