polyend/TrackerBetaTesting #649: Allow playback while in Sample Recorder

0x0zzz, 2021-01-07:
Currently, while in Sample Recorder, pressing the Play button does nothing.

Starting playback in this mode by pressing the Play button would help with recording performances played on external instruments through Line In along with the current pattern you are working with (i.e. acoustic and other instruments without MIDI).

An alternative to this could possibly be a metronome in the Sample Recorder, if there is some technical reason why this isn't feasible.

synthexpert, 2021-01-09:
This would be a great feature!

gurkenkimchi, 2021-01-20:
would be also great if you return back to the sample recorder page after recording line in saving and loading a sample rather than a jump to the instruments page.

Goldfasan, 2021-11-12:
Considering the competition (4o4mk2), this would be a high value feature... This alone is a decision maker, simple and ingenious.

closed 2021-12-02 by Jacknametrouble