polyend/TrackerBetaTesting #656: Suggestion --> Better/faster scrolling of the main encoder

anemlife, 2021-01-08:
The speed of the main knob feels really slow, especially when adjusting samples points (start/end, granular position, length, etc...). I like this big knob but I wish it was more precise, when trying to go fast it sometimes even doesn't move! Not sure if this is a hardware/software limitation or if this could be improved? 
I remember you already did a few improvements in previous firmwares but it still feels a bit rude in 1.3.0, that's why I'm doing a new topic about it!

I would love if the PT could kind of detect when we do an "acceleration" in scrolling, and make the moving point go way faster than when do slow/precise adjustment. That would be really useful also in Sample Editor, at the moment it feels pretty long to adjust things with this slow scrolling which is annoying compared to the fast workflow of this machine! 

A better feeling response curve would be really appreciated! Thanks :)

swindus, 2021-01-19:
Yes, got the same 'problem'. Using the knob is sometimes really wonky because the values changes get stuck or moving to fast. I think it would make sense to work on this to optimize the response of the knob. The knob is one of the center controls of Tracker and it should work smooth and precise at the same time.

doelman, 2021-01-19:
I agree, the knob is annoying and slowing me down me very often, especially with selecting start /end times of samples it could take ages and it's kind of unresponsive as well at some points.

SenpaiAndre, 2021-02-17:
Also having issues with the encoder getting stuck or having settings roll/jitter backwards when trying to adjust options quickly. 
I'm not sure if it's a hardware or software issue at this point, luckily I haven't seen many complaints about it.

https://user-images.githubusercontent.com/49768918/108142121-3e672400-7119-11eb-8643-5017e0381645.mp4

A quick solution would be to add adjustment scale modifiers for the encoder and arrow switches: Holding shift while turning the encoder or using the arrow keys could multiply the step/position amount by a assignable parameter in the tracker config. (2x, 5x, 10x, etc)

swindus, 2021-02-17:
Thanks for the video. This is how it (not) works for me too.

anemlife, 2021-02-17:
@SenpaiAndre 

I'm surprised about how buggy it is in your video! I have little issues with mine but not as much as yours. Just curious, what firmware version are you using?

swindus, 2021-02-17:
Here is my encoder. It not really usable at all ....

https://user-images.githubusercontent.com/6263942/108218163-abd47e00-7134-11eb-87e6-248550124eae.mp4

swindus, 2021-02-17:
I'm using the firmware version 1.3.1b3. Is it a broken encoder???

muhfugga, 2021-02-17:
I have the same issue

SenpaiAndre, 2021-03-01:
> @SenpaiAndre
> 
> I'm surprised about how buggy it is in your video! I have little issues with mine but not as much as yours. Just curious, what firmware version are you using?

1.3.1 b2 both at the time of recording and currently

Jacknametrouble, 2021-03-02:
Please submit a video clip showin the issue to https://help.polyend.com/submit_ticket

closed 2021-03-02 by Jacknametrouble