polyend/TrackerBetaTesting #659: [1.3.1b1] Sample of deleted note keeps looping

aheger, 2021-01-08:
### Steps to reproduce

- Create Instrument with Forward, Backward or Pingpong Loop
- Fill pattern with notes
- Play the pattern
- Delete the notes

**Expect:**
Silence

**Actual:**
Sample keeps looping

### Attachments and links

https://user-images.githubusercontent.com/55574996/104071735-0dd2c700-520a-11eb-8725-e55f21079963.mp4

### Other important information

* How often does this bug happen? every time

* Does this bug happen in previous firmware versions? don't know

* Does this bug happen in every project? yes

modulationsource, 2021-01-08:
I would not expect silence.
There is no Note off, so why should the sound stop ?
Deleting the notes just means, there are no notes that can be triggered.
So deleting the notes leads to the same behaviour like changing to an empty pattern.

aheger, 2021-01-08:
Mmh yes that makes sense. Thank you.

closed 2021-01-08 by aheger