polyend/TrackerBetaTesting #662: [1.3.0 & 1.3.1b1] Constantly crashing in song mode

benc-uk, 2021-01-09:
The device crashes (hard lock, requiring power cable removed) constantly in song mode.  
The Tracker is unusable.

### Steps to reproduce

Work in song mode for between 2 and 15 min, and the device will lock up requiring power cable to be removed. When you power back up the problem will reoccur.

### Attachments and links

The behaviour is not deterministic so it's not easily reproducible, but only happens in song mode, but *always* happens. Sometimes it's during a copy & paste sometimes when switching back to a different mode, sometimes it's nothing and it's just frozen when you looked away for a moment.

### Other important information

* How often does this bug happen? Every time working in song mode for more than 5 minutes, but actual time varies

* Does this bug happen in previous firmware versions? 1.3.0 and 1.3.1b1

* Does this bug happen in every project? Yes, even new ones.

adfielding, 2021-01-09:
Can confirm, I've only had the Tracker for a few days and I've had this issue several (at least ) times. I've only been using the Tracker with 1.3.0 and 1.3.1b1 and it always seems to occur from song mode while audio is playing.

I've tried using both mains power and power from a USB port on my computer. Every time it's happened I've had to perform a hard reset, but I've (thankfully) never lost any data as a result of it.

ambv, 2021-01-09:
Are you using MIDI at all, @adfielding, @benc-uk?

adfielding, 2021-01-09:
> Are you using MIDI at all, @adfielding, @benc-uk?

No, I've only used it stand-alone so far.

benc-uk, 2021-01-09:
I am using external MIDI, via the MIDI jack not USB. Just notes and CC, not sending clock or transport

aheger, 2021-01-10:
Can confirm the issue, though I have no reproducer.

Just had similar crash on 1.3.1b2 when just starting a new project.
2 instruments, 1 track, no midi.
Device's MIDI IN was connected to a Novation Launchpad MK3 via USB through a Raspberry PI using `aconnect`. 
Played around with instrument automation while the only pattern (**not** song mode) was playing, suddenly device froze and loud high pitched sound was played, had to pull the USB cable.
Could not reproduce after reboot.

Find project here:
[crash.zip](https://github.com/polyend/TrackerIssues/files/5792056/crash.zip)

Also here is the `trackerLog`:
[log.zip](https://github.com/polyend/TrackerIssues/files/5792057/log.zip)
Though it does not seem to contain anything useful besides 19k entries of `changed pattern`.

benc-uk, 2021-01-12:
Done some more testing / investigation

- I tried moving my project to a different SD card in case that was the cause - it still crashed after a few minutes. The SD card is not the cause.
- I created a new totally project on this SD card, and used only the samples that were provided with the Tracker (so no samples of my own or anything sampled in). For a long time it didn't crash. Then suddenly when I switched from song mode to the master view it crashed

I'd say most (but definitely not all) crashes happen when switching from song mode to another mode

Attached are both projects, both are unfinished
[test.zip](https://github.com/polyend/TrackerIssues/files/5804136/test.zip)
[WTF.zip](https://github.com/polyend/TrackerIssues/files/5804137/WTF.zip)

closed 2021-01-23 by ambv