polyend/TrackerBetaTesting #668: Multiple Sidechain Sources

nakajima, 2021-01-09:
Hello! It would be great if we could choose multiple tracks for sidechaining the limiter. There are times that I want multiple sounds to take emphasis at once and that’s difficult with only one track being the sidechain source. I also sometimes want to have my kick and snare on multiple tracks for mixing purposes but still sidechain to both.

Thanks!

anemlife, 2021-01-09:
In addition to this, what if we could make the sidechain linked to specific samples, not only tracks? This would allow for sidechaining without having to spend a whole track as your sound source! 

let me explain: at the moment, I use a single track for all my kicks but then I can't put other samples in this track as they will affect the sidechain! Using samples as the source instead of tracks could avoid this problem :)

LetsSeeIfWeSolveThis, 2021-01-11:
#577 duplicated

closed 2021-12-02 by Jacknametrouble