polyend/TrackerBetaTesting #671: Question (1.3.0) : Clock Through behaviour / BPM Display

tilllt, 2021-01-09:
This is not so much of a feature request but more of a question:

Right now i have my Computer + DAW ---USB---> Tracker ----MIDI----> Korg Volca Beats. The Clock goes through nicely from Computer to Volca Beats while being in Play Mode on the Tracker, Speed changes make it through to the Volca Beats. 
Tracker Config is: 
Clock In: USB
Clock Out: MIDI Out Jack

Once i stop the tracker, the clock is no longer passed through and the Volca changes speed to its internal setting. Is this intended behaviour? i would expect the clock signal of the DAW to be passed through no matter what, the way its set up right now, even when the Tracker is not playing itself...

Additional Question: When receiving external Clock, is there a quick way to see the incoming BPM? It just says "EXT" where normally the BPM would be noted.

tilllt, 2021-01-10:
Can anyone comment how this is handled on other devices that have USB Clock In and MIDI Clock OUT?

pawal, 2021-01-11:
I believe that the MIDI standard says that the clock is to be running at all times (and this is what i.e. Elektron implements). But all vendors are behaving differently. I would say in this case that the Volca behaves a little bit strange by changing its clock setting. Other devices I have tries to keep the external average tempo itself, when a clock is no longer sent.

tilllt, 2021-01-11:
Well, technically i am using the Tracker as a Clock converter between USB and MIDI, so it just passed the external Clock from the DAW through to the Volca. I just dont know if its supposed to stop doing that, when the Tracker is not running itself. In my limited understanding of the matter, the external clock should be passed through to other devices, if its running or not.