polyend/TrackerBetaTesting #675: Song Mode - Faster Pattern Selection

stustumaru, 2021-01-11:
First off, I absolutely love the Tracker.

One thing I've noticed is that changing patterns in song mode could be sped up a bit. As it currently stands, the only way to select a pattern is to click the << or >> button at the bottom. If I've built a track from the ground up in order, it's fast because the next pattern is usually only one or two away from the current pattern, but it can take quite a while when I've made a lot of different sections and am building the song up from scratch. Changing the number from pattern 20 to pattern 1, for instance, takes 19 button presses, and that can add up a lot of extra time over the course of building a single song.

**My proposal is to allow the jog wheel to scroll through pattern numbers in song mode when either the << or >> button is held. I feel like that would be intuitive and match the existing UI, and it would greatly speed up song creation.**

Great work on this amazing instrument! I have a big eurorack setup and multiple other synths/drum machines (not to mention Ableton Live, etc.), but I've made probably 90% of my music on the Tracker since getting it in October. Lovely stuff.

ConnorCarbon, 2021-01-27:
Can confirm as a new PT user that I have tried to turn the jog wheel multiple times in song mode to select a different pattern utilized by a slot in song mode and had to pause to realize I have to press the << >> buttons

clickysteve, 2021-02-21:
+1 on this, though I would suggest utilising the UI/UX from elsewhere, to allow faster selection of specific patterns via a pop up. Having to scroll through individual patterns in song mode is a real drag. Want to get to pattern 255? That takes 254 key presses! That's nuts.