polyend/TrackerBetaTesting #680: [1.3.1b2] Performance mode isse

phaelam, 2021-01-14:
when I play the Song I have laid out, and go into Performance mode, it changes playback to Loop.
should it not continue playing the song?

tfspsound, 2021-01-17:
I’d like to know what polyend make of this issue. Will / can it be fixed or do they even intend to? I’ve seen a couple of threads on this, I submitted one myself #413 but they never seem to gain any following from other users. I can’t speak for everyone but I think the performance mode is a big part of what makes the tracker different from other trackers and groove boxes, however I also find the performance mode becomes a little redundant without the ability to effect my completed songs. I love the tracker but this issue is holding it back.

phaelam, 2021-01-17:
Sorry about the double up. I don’t seem to ask the right questions in the search to find a common thread. 

I agree with you. Using Performance mode on a looping part of the pattern doesn’t make sense. Even if it has to be in Pattern mode to werk, it would be nice if the Song continued to play.

tfspsound, 2021-01-17:
No worries, I’ve done the same thing, we could do with some sub categories like Pattern Mode, Song Mode, Performance Mode and Step FX instead of just Bugs and Enhancements. 
I’d keep yours as well it’s the most recent suggestion for this alteration and hopefully other new users can get on board too.

ambv, 2021-01-19:
Duplicate of #413, let's discuss there.

closed 2021-01-19 by ambv