polyend/TrackerBetaTesting #683: Insert empty pattern in between patterns

OnevenHub, 2021-01-14:
Often when working on a track I find I want to add a new pattern after the previous, but that the next pattern  is already filled up with information. 

Instead of going through each manually and copy/pasting patterns(you now have to manually 'move up' your whole track, it would be great if there was a command that would simply create a new blank pattern in between existing patterns. Would be a great workflow improvement.

MetabaronCZ, 2021-01-14:
Yes, I would also use that feature. Thank you for posting this request.

sunfallz, 2021-01-16:
Would be a great feature :)

jonwaterschoot, 2022-05-19:
Cross posting from the[ lines topic](https://llllllll.co/t/polyend-tracker/30124/703?u=jonwtr ): - adding a link to this request there as well

> _Is there a way to insert a new blank pattern in between already existing patterns?_

> I don’t believe this is possible now.
> A way to reorganize patterns would indeed be very nice.
> My approach is mainly to use the top row for the first ideas, iterations or alternative variations I put on the third and fourth row.
> If it would be possible to select multiple patterns, copy and move them, I agree, this would be cool.
> Of course implementing this would be tricky for the song mode. And who knows what other errors.

ryancara, 2022-05-19:
I would also love this!