polyend/TrackerBetaTesting #684: Higher sound slot count

OnevenHub, 2021-01-14:
With the low quality import being added I'm hoping you can also increase the sound slots from 48 to 96. Especially when making a live set the maximum of 48 is limiting unless you want to use the same sounds over and over again.

ketumatek, 2021-01-14:
For live performance only 48 samples per projet is really small..(and no possibility to play a projet after another without stop the machine),if your are working with sample and you want a really difference of sound between all your tracks or projet.
So double it to 98 will be more easy for a live set...
But not sur it will be possible because of memory.

MetabaronCZ, 2021-01-14:
I always thought, that trackers are used to make game music. You are using it for whole live set performance? Interresting.

OnevenHub, 2021-01-14:
Lol, you have +200 patterns for a reason right. It's light, compact so ideal for live performing. Powering it from a power bank allows for nice last minute adjustments while travelling too.

ketumatek, 2021-01-14:
> I always thought, that trackers are used to make game music. You are using it for whole live set performance? Interresting.

Not yet but I would like to...

OnevenHub, 2021-01-15:
> > I always thought, that trackers are used to make game music. You are using it for whole live set performance? Interresting.
> 
> Not yet but I would like to...

Definitely fun! Making an hour set was a great way for me to get to learn tracker.

ArtFluids, 2021-03-23:
I was going to come here to suggest this.
I would love a much higher instrument count. I want to do projects with a lot of really small samples. RAM isn't the issue here, it's just the limitation of only having 48 instruments. I don't need more room for audio, just more instrument slots.

Like I want to do fun patterns where I sample a bass synth at various filter cutoffs for acid basslines, but I will quickly run out of instrument slots just for that one sound. If that one sound uses 12 different samples, that's 1/4 of my available instrument slots just for that one idea.

I guess I could always use slices to sneak in more sounds per-instrument but that's not easy when I want to swap out some of the sounds. 

I know that the number of instruments corresponds to the number of pads, but it would be nice to be able to scroll beyond 48. I would love like at least double that or even higher.

static-noise-PT, 2021-04-20:
> I was going to come here to suggest this.
> I would love a much higher instrument count. I want to do projects with a lot of really small samples. RAM isn't the issue here, it's just the limitation of [...]

I see it similar but I also see the complex of "only" having 48 pads. This means that we would need to have 2 banks a 48 and swap via pages. I don't know how reliable / user friendly this would be.

And I don't see any issue with having a performance. Just a mind game:
10 Track-Set:
Drums -> 2 Samples via Samplechains (ofc it depends what you perform but I usually perform my own music mostly EP/LPs)
Bass -> 10 Samples
Pads, Leads, etc. -> 20 Samples
Left Slots / Sounds -> 16 Samples left