polyend/TrackerBetaTesting #690: [1.3.1.b2] Small weird behavior; command (shift+up) after Shift+Home) is not working properly

LetsSeeIfWeSolveThis, 2021-01-14:
1. Staying at any step of the track (but not the first).
2. Push **_SHIFT+HOME_** and then **_SHIFT+UP_**, 
3. The Tracker **_will select directly all Tracks_** and not that single track on which you are situated.

The previous behavior was **SHIFT+double UP** to select the eight tracks.

It doesn't happens every time, but it is usual, try keeping holding **SHIFT** while changing from **HOME** to **UP**.

bambrose, 2021-01-17:
Yeah I just noticed this. Real workflow killer. Shift + up selects all rows instead of just the row you're in.
Bloody annoying

ambv, 2021-01-19:
I'll be looking into this but testing just now I cannot reproduce this yet.

talktogreg, 2021-01-19:
Tested myself. 

Have worked out how to reproduce. If, after Shift+Home, you release Home then click Up (or release Shift+Home and click Shift+Up) then it works normally.  But in either case, if you hold the Up (or Shift-Up) for more than 0.5 seconds, it highlights all columns.

In fact, you can start anywhere with nothing selected, and Shift-Up will (if held) go from selecting step above, to entire columns, to all columns.

bambrose, 2021-01-19:
I wonder whether it is related to the 'arm track' option that was introduced in 1.3 (or thereaboits) I use that a lot (i.e. restrict recording to 2-3 tracks rather than all 8)
When exiting that mode is when I noticed the 'broken' shift + up behaviour

bambrose, 2021-01-20:
I'm not getting this with my newer mk2 machine... only on my mk1 ?!
I also note that the scroll wheel is slower on the newer mk2 machine. I think that may be where the issues lies

bambrose, 2021-01-21:
Edit : sorry. Bum Steer,
Am now seeing the same behaviour with Mk2
Appreciate the above is less than stellar bug reporting. sorry.
When I do get the issue.. the length of press on the buttons is not having any effect

talktogreg, 2021-02-22:
Hi (again),

Have done some further personal testing and found another way to make this BUG occur (on top of the HOLD action).

It appears that this (highlighting all track columns at once) occurs if LIVE-REC has recently been started and subsequently stopped. If sequencer is then stopped and REC (red mode) enabled, then the BUG frequently occurs regardless of how long buttons are held.

bambrose, 2021-02-22:
That makes sense and tallies with what I find. It seems though that this use of Live+Rec then 'corrupts' the project such that the behaviour being described is evident every time the same project is re-opened.

closed 2021-03-11 by ambv

bambrose, 2021-03-11:
@ambv you fixed it :)
Bravo !! Nice Job