polyend/TrackerBetaTesting #694: Quicker Settings editing

talktogreg, 2021-01-15:
Whilst in CONFIG menu, you can navigate through the columns using the arrow keys. This includes being able to use the right-arrow to enter a value settings column. (ie Time Scale Denominator).

However, you have to use the 'ACCEPT' screen key to select a value.

Could a further right-arrow on 'selected' option mimic the 'ACCEPT' screen key?

keith-macleod, 2021-01-15:
It would be cool if the home/insert key worked everywhere as an enter key like it does in the sample loader.
Another thing that would make it a smoother experience, eliminating menus where they're not particularly necessary, like toggle values, or just shift+jogwheel to edit values directly.

tfspsound, 2021-01-17:
I wish Shift + Jogwheel had been made into greater increments on whatever value was being manipulated but doesn’t it effect octave now? I think holding Note + Left / Right would have made a better octave option similar to how you’d input note data on the game boy. I agree a unified enter button would be useful and the elimination of less useful buttons like Stop on the Song screen and replaced with something useful like a Loop Selection option. I’d also like to see the Pattern and Step screen buttons in Pattern mode become latching switches.

x2db, 2021-04-14:
Shift+jog would be a life savior in: 
a) Sample editing (moving start and stop points)
b) Most of the Envelopes editing (moving attack, sustain to 0 etc)
c) Most of the Sample Effects (any frequency adjustments is a pain, try to use the WT enhancer   )

eingedanke, 2021-06-22:
Yes I was actually surprised when I noticed that the insert key was not working as "enter" everywhere.
Same with the "Home" key, why is it not consistently taking you to the top of a list, e.g. in "Config", "Sample Loader", etc. ?

And "Shift + Jog Wheel" sounds very useful too, especially in the sample editor.