polyend/TrackerBetaTesting #701: [1.3.0] In Pattern Note Preview is different from when played

sunegg1024, 2021-01-15:
## Hi, sorry you had trouble!

### Steps to reproduce

Rolls, Filter modulation and other effects don't seem to preview properly when Previewe'd in pattern screen. They sound different when played in the full track. I can understand if this is due to LFOs not being triggered yet for Filter and Volume but the rolls would be nice to be able to preview without playing the pattern.

### Attachments and links

Can send a video, if needed.

### Other important information

* How often does this bug happen? (every time, often, "it only happened once") **Every time**

* Does this bug happen in previous firmware versions? **Not test**

* Does this bug happen in every project? **Yes**

thebradclay, 2021-01-17:
For rolls, highlight the step where the roll begins through where it ends. Hit preview. You should hear the roll. If you are only highlighting the step where the roll fx is placed and nothing after it won't play anything in preview mode except for the highlighted step. As for the t fx I don't know off the top of my head.

MortalJS, 2022-03-19:
No response. Closed

closed 2022-03-19 by MortalJS