polyend/TrackerBetaTesting #704: Instant pattern change via external CCs

PEtajaja, 2021-01-16:
I really like that you added the ability to have some control of the tracker externally, I was hoping that I would be able to trigger an instant pattern switch rather than a sequential one, I think if there was a choice between the two, instant switch would be more useful as you'd still be able to achieve the same results for the most part.

 So my suggestion:
 
 Either add an option in the MIDI config to set up CC 41 - 48 to either trigger a sequential change or an instant change
 or
 Replace the current behavior with instant pattern change
 or
 Maybe there's some other way of implementing that I'm overlooking!
 
This would be such an incredible powerful feature in that it'd allow us to shuffle various patterns together by sending a sequence of CC 41- 48 to the tracker.

sunfallz, 2021-01-16:
Agreed. This would be a fantastic enhancement.

PEtajaja, 2021-05-03:
Any chance we could see this in an update? It'd be an incredible enhancement

philvoris, 2021-07-31:
Also keen on this.