polyend/TrackerBetaTesting #707: Looper Performance FX

tfspsound, 2021-01-17:
This differs from Step Repeat in that it would require the tracker to continuously monitor the master out and when the Looper is triggered the master out is recorded for the duration set in one of the three setting slots, then looped until cancelled or when another slot is triggered. I guess it’s be like ram record on a machinedrum. Don’t know if it’s possible on the tracker but I think it could be useful if it is.

ambv, 2021-01-19:
Sounds like a fun idea but we're unable to do it hardware-wise due to insufficient memory.

closed 2021-01-19 by ambv

tfspsound, 2021-01-19:
Curses. Oh well. If you ever manage to find a few spare mb just enough for 8 to 12 steps would do. Cheers